gfvalvo/NewEncoder #3: First turn after switching direction is not recognized

vedymin, 2020-09-01:
Using example code for single encoder, in FULL_PULSE.
Encoder hardware is Waveshare 9533.

Steps to repeat:
* Use example code from NewEncoder/examples/SingleEncoder/SingleEncoder.ino
* Use FULL_PULSE
* Open serial monitor and wait for:
`
Starting
Encoder Successfully Started at value = 0
`
* Turn encoder clockwise

Expected:
`
Starting
Encoder Successfully Started at value = 0
Encoder: -1
`

Actual:
`
Starting
Encoder Successfully Started at value = 0
`

* Turn encoder clockwise
Expected:
`
Starting
Encoder Successfully Started at value = 0
Encoder: -1
Encoder: -2
`

Actual:
`
Starting
Encoder Successfully Started at value = 0
Encoder: -1
`

* Turn encoder counterclockwise
Expected:
`
Starting
Encoder Successfully Started at value = 0
Encoder: -1
Encoder: -2
Encoder: -1
`

Actual:
`
Starting
Encoder Successfully Started at value = 0
Encoder: -1
`

In simple words, if you turn left right left right left right  then nothing happens. Counter is changing value only with second and above turn in one direction. If you switch direction, then first turn in the new direction is not changing counter value.

gfvalvo, 2020-12-18:
Not reproducible on any encoders or processors I have available.

gfvalvo, 2020-12-18:
Does the Waveshare 9533 one complete quadrature cycle every detent or one complete quadrature cycle for every two detents?

closed 2020-12-21 by gfvalvo