gfvalvo/NewEncoder #10: Min-Max question

SteveRMann, 2020-12-18:
How do I set no minimum or no maximum?

gfvalvo, 2020-12-18:
You can't. You can set minValue to its minimum value (-32768) and maxValue to its maximum value (32767). Allowing the count to continue beyond those limits would result in signed integer overflow and undefined behavior.

SteveRMann, 2020-12-19:
"Allowing the count to continue beyond those limits would result in signed integer type overflow/underflow which produces undefined behavior."

I disagree.

That should be my decision.  There is nothing in your code that would be bad should a rollover happen.  It should be the user's decision to handle or ignore a rollover.

gfvalvo, 2020-12-19:
You are, of course, welcome to fork the code from GitHub and implement whatever functionality you like. That's the beauty and power of the Open Source paradigm. However, I will not be implementing potentially undefined behavior in my code.

gfvalvo, 2020-12-19:
OK, I've added a virtual updateValue() function that will allow you to create new versions of the class via inheritance. You can use this function to customize the increment / decrement and min / max behavior. See the 'CustomEncoder' example.

closed 2020-12-21 by gfvalvo