gfvalvo/NewEncoder #11: is there a loop back function?

sebasdt, 2020-12-18:
is there a way or function that when the max encoder value is reached it will loop back to the min value or the other way around?

for example min value = 0 and max value = 5.
it should go something like this:
0-1-2-3-4-5 -> 0-1-2-3-4-5 (clockwise)
5-4-3-2-1-0 -> 5-4-3-2-1-0 (counter clockwise)

BTW the debouncing of the outputs from the encoder is so well done! no weirdness or missing turns at all.

gfvalvo, 2020-12-18:
No, I didn't think it made sense to implement that feature. If you're using the encoder to control something (like audio volume or an object's physical position), a wraparound would cause a gigantic jump in the setting that could be problematic.

If you like, you could implement it in your code using the getValue(), setValue(), upClick(), and downClick() methods.

sebasdt, 2020-12-19:
thanks for the tip.
if this function is added you could make it so the function could be turned on or off.
example: setBoundaries(0,  300, true);  true = looparound enabled
                                      min-max bool

sebasdt, 2020-12-19:
Thanks for the tip!

If you wanted to add this feature you could do it so that it could be enabled and disabled.
Example: setBoundaries(0,  300, true)----> min value, max value and a bool (true = loop around enabled).

gfvalvo, 2020-12-19:
OK, I've added a virtual updateValue() function that will allow you to create new versions of the class via inheritance. You can use this function to customize the increment / decrement and min / max behavior. See the 'CustomEncoder' example.

sebasdt, 2020-12-20:
thanks for the help so far! I will take a look.

closed 2020-12-20 by sebasdt