gfvalvo/NewEncoder #15: adding the most simple demo ever possible

StefanL38, 2021-11-23:
Hi gfvalvo,

I have looked through the example-codes. IMHO Even the singleEncoder is not simple

With te NewEncder-version 2.0:
is it still possible to have it as simple as these three lines?

NewEncoder myEncoder(13, 15, -20, 20, 0, FULL_PULSE);
myEncoder.begin(parameters)
value = myEncoder

?
best regards Stefan

gfvalvo, 2021-11-25:
It is. But you'll get a compiler warning stating that function is deprecated and may be removed in a future release. It's all documented in the Version 2.x README.

StefanL38, 2021-11-27:
I mean something like this

less lines of code but more explanation inside of the code through comments
[code]
#include "Arduino.h"
#include "NewEncoder.h"

// Pins 2 and 3 should work for many processors, including Uno. 
// explanation:

// NewEncoder myEncoder(2, 3, -20, 20, 0, FULL_PULSE);

// create an encoder-object called "myEncoder"
// "2" IO-pin for first channel of rotary-encoder
// "3" IO-pin for second channel of rotary-encoder
// "-20" lower limit to count down to
// "20"  upper limit to count up to
// "FULL_PULSE" type of pulse (other option HALF_PULSE)
// Use FULL_PULSE for encoders that produce one complete quadrature pulse per detnet, such as: https://www.adafruit.com/product/377
// Use HALF_PULSE for endoders that produce one complete quadrature pulse for every two detents, such as: https://www.mouser.com/ProductDetail/alps/ec11e15244g1/?qs=YMSFtX0bdJDiV4LBO61anw==&countrycode=US&currencycode=USD
// 
NewEncoder myEncoder(2, 3, -20, 20, 0, FULL_PULSE);

int value = 0;
int lastValue = -1;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");

  if (!myEncoder.begin()) {
    Serial.println("Encoder Failed to Start. Check pin assignments and available interrupts. Aborting.");
    while (1) {
      yield();
    }
  } else {
      Serial.print("Encoder Successfully Started at value = ");
      Serial.println(myEncoder);
  }
}

void checkEncoder() {
  value = myEncoder;

  if(value != lastValue) {
    Serial.print("Encoder value= ");
    Serial.println(value);
    lastValue = value;    
  }
}

void loop() {
  checkEncoder();
}

// the behaviour of the code that you can see in the serial monitor is
// the encoder-vale starts with 0 and depending on the rotating-direction
// counts up until 20 but not higher or counts down to -20 but not lower 
// so in the serial monitor you should see
/*
Starting
Encoder Successfully Started at value = 0
Encoder value= 0
Encoder value= 1
Encoder value= 2...counting up
Encoder value= 20
Encoder value= 19
Encoder value= 18...counting down
Encoder value= 2
Encoder value= 1
Encoder value= 0
Encoder value= -1
Encoder value= -2... counting down
Encoder value= -19
Encoder value= -20
*/
[/code]

StefanL38, 2021-11-27:
this is a test-comment to learn how to insert code as a code-section in github

`#include thecode.h`

gfvalvo, 2021-11-27:
You can write your code that way and it will work for now. But, as noted in the README, you're using deprecated functions and will get compiler warnings. So, I don't recommend using them. And I'll probably remove them in a future release.
```
sketch_nov27a: In function 'void setup()':
sketch_nov27a:34: warning: 'NewEncoder::operator int16_t() const' is deprecated: May be removed in future release. See README and library examples. 
       Serial.println(myEncoder);
                               ^
In file included from C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_643818\sketch_nov27a.ino:2:0:
D:\Arduino-1.8.15\portable\sketchbook\libraries\NewEncoder/NewEncoder.h:314:2: note: declared here
  operator int16_t() const;
  ^
sketch_nov27a: In function 'void checkEncoder()':
sketch_nov27a:39: warning: 'NewEncoder::operator int16_t() const' is deprecated: May be removed in future release. See README and library examples. 
   value = myEncoder;
         ^
In file included from C:\Users\GFV\AppData\Local\Temp\arduino_modified_sketch_643818\sketch_nov27a.ino:2:0:
D:\Arduino-1.8.15\portable\sketchbook\libraries\NewEncoder/NewEncoder.h:314:2: note: declared here
  operator int16_t() const;
  ^
```

gfvalvo, 2021-11-30:
Having recommended to @StefanL38 that the deprecated functions not be used, I'm closing this issue.

closed 2021-11-30 by gfvalvo