gfvalvo/NewEncoder #18: (currentValue); gives -24287 at bootup

Potatopasty, 2022-01-29:
Hi! 

I have a weird issue (weird to me) with my latest project. After reset I get (currentValue); of -24287 on my oled. After I rotate the encoder it gives 101 or 99 which is correct, and only then timer starts and everything works normally. However I can't get this problem to show on serial monitor. What am I doing wrong?

Code here: https://github.com/Potatopasty/Synth-master-clock/blob/main/Master_Clock.ino

Thank you for the great library.

gfvalvo, 2022-01-30:
This is more than likely a problem with your code and not the NewEncoder library.

One problem that I see is the first time 'cycle_on()' is called, the global variable 'BPM' will have a value of zero. Yet, you divide by it:
`int cycletime = (60000 / BPM);`
That will cause undefined behavior.

Also, the SimpleTimer function 'setTimeout()' takes 'long' for the interval, not a 'float'

Also, you never initialize the variable 'prevEncoderValue'.

If you still believe this is an issue with the library, please post a simple [MRE](https://stackoverflow.com/help/minimal-reproducible-example) that demonstrates the problem.

Potatopasty, 2022-01-30:
Thank you for excellent answer and your time! It was indeed the problem with BPM having a value of zero at boot. With your help I managed to hammer in a solution. Not surprisingly issue was with me, and not the library.

closed 2022-01-30 by Potatopasty