gfvalvo/NewEncoder #20: How can i add support for stm32? (stm32f103rct6)

markonjak, 2022-03-09:
Hello,

i just want to know how i can add stm32 microcontroller support (stm32f103rct6), not for all pins just 2 is enough, would really appreciate guidance on how to implement this. Tried adding interrupt pins definitions but it doesn't compile, not sure what exactly to add in direct_pin_read header.

what i tried adding:

interrupt_pins.h:
`#elif defined(genericSTM32F103RC)

#define CORE_NUM_INTERRUPT 2
#define CORE_INT0_PIN 37
#define CORE_INT1_PIN 38
`

direct_pin_read.h:
`#if defined(genericSTM32F103RC)

#define IO_REG_TYPE			uint32_t
#define PIN_TO_BASEREG(pin)             (portInputRegister(digitalPinToPort(pin)))
#define PIN_TO_BITMASK(pin)             (digitalPinToBitMask(pin))
#define DIRECT_PIN_READ(base, mask)     (((*(base)) & (mask
`

Best regards.

gfvalvo, 2022-03-10:
I've never used an STM32 nor do I own a board with one to test it.

The code you posted is incomplete, so I have no way of knowing what you did wrong. If you were to supply the complete code for the file you modified, perhaps I could point you in the right direction.

markonjak, 2022-03-10:
What i posted in the top message is all i added.
I think the problem is because i defined the board name wrong?

interrupt_pins.h
```
// interrupt pins for known boards

// Teensy (and maybe others) define these automatically
#if !defined(CORE_NUM_INTERRUPT)

// Wiring boards
#if defined(WIRING)
  #define CORE_NUM_INTERRUPT	NUM_EXTERNAL_INTERRUPTS
  #if NUM_EXTERNAL_INTERRUPTS > 0
  #define CORE_INT0_PIN		EI0
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 1
  #define CORE_INT1_PIN		EI1
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 2
  #define CORE_INT2_PIN		EI2
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 3
  #define CORE_INT3_PIN		EI3
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 4
  #define CORE_INT4_PIN		EI4
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 5
  #define CORE_INT5_PIN		EI5
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 6
  #define CORE_INT6_PIN		EI6
  #endif
  #if NUM_EXTERNAL_INTERRUPTS > 7
  #define CORE_INT7_PIN		EI7
  #endif

// Arduino Uno, Duemilanove, Diecimila, LilyPad, Mini, Fio, etc...
#elif defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__) ||defined(__AVR_ATmega168__) || defined(__AVR_ATmega8__)
  #define CORE_NUM_INTERRUPT	2
  #define CORE_INT0_PIN		2
  #define CORE_INT1_PIN		3

// Arduino Mega
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
  #define CORE_NUM_INTERRUPT	6
  #define CORE_INT0_PIN		2
  #define CORE_INT1_PIN		3
  #define CORE_INT2_PIN		21
  #define CORE_INT3_PIN		20
  #define CORE_INT4_PIN		19
  #define CORE_INT5_PIN		18

// Arduino Nano Every, Uno R2 Wifi
#elif defined(__AVR_ATmega4809__)
  #define CORE_NUM_INTERRUPT	13
  #define CORE_INT0_PIN		0
  #define CORE_INT1_PIN		1
  #define CORE_INT2_PIN		2
  #define CORE_INT3_PIN		3
  #define CORE_INT4_PIN		4
  #define CORE_INT5_PIN		5
  #define CORE_INT6_PIN		6
  #define CORE_INT7_PIN		7
  #define CORE_INT8_PIN		8
  #define CORE_INT9_PIN		9
  #define CORE_INT10_PIN	10
  #define CORE_INT11_PIN	11
  #define CORE_INT12_PIN	12
[... 386 lines cut ...]

```

I used this core https://github.com/stm32duino/Arduino_Core_STM32
After some further digging i got it to compile by adding this.

direct_pin_read.h
```
#elif defined(STM32F103xE)

#define IO_REG_TYPE			uint32_t
#define PIN_TO_BASEREG(pin)             (portInputRegister(digitalPinToPort(pin)))
#define PIN_TO_BITMASK(pin)             (digitalPinToBitMask(pin))
#define DIRECT_PIN_READ(base, mask)     (((*(base)) & (mask)) ? 1 : 0)
```
interrupt_pins.h
```
#elif defined(STM32F103xE)

#define CORE_NUM_INTERRUPT 2
#define CORE_INT0_PIN PC6
#define CORE_INT1_PIN PC7
```
But i don't know if these parameters are ok, will have to test it.
Btw what does this line do `#define DIRECT_PIN_READ(base, mask)     (((*(base)) & (mask)) ? 1 : 0)` ?

gfvalvo, 2022-03-15:
> But i don't know if these parameters are ok, will have to test it.

gfvalvo, 2022-03-15:
> But i don't know if these parameters are ok, will have to test it.

Yes, I've never used STM32 and am not familiar with the architecture.

> Btw what does this line do #define DIRECT_PIN_READ(base, mask) (((*(base)) & (mask)) ? 1 : 0) ?

That's defining a macro function for reading a pin's status. In most architectures it's much faster than Arduino's digitalRead() function. It uses the base address for the particular GPIO register (supplied by PIN_TO_BASEREG) and the bit position mask (supplied by PIN_TO_BITMASK)

gfvalvo, 2022-12-14:
I've added experimental STM32 support to the library. I still don't have an STM32-based board, so it's untested. Give it a try if you get a chance and let me know.

kakhi-kotchauri, 2022-12-16:
Everything works perfectly except I can't figure out how to change increment / decrement values, library only increases / decreases counter by one, but I want to increase by lets say 25, also I don't want any dynamic changes I just want to set the increment to 25, by the way English isn't my native language so please don't judge me

gfvalvo, 2022-12-18:
I understand from you other issue entry that the STM32 support seems to be working. I'll address your other question in that issue.

gfvalvo, 2023-01-27:
Have not heard back from [markonjak](https://github.com/markonjak). Issue closed.

closed 2023-01-30 by gfvalvo