gfvalvo/NewEncoder #22: change increment / decrement values

kakhi-kotchauri, 2022-12-17:
Everything works perfectly except I can't figure out how to change increment / decrement values, library only increases / decreases counter by one, but I want to increase by lets say 25, also I don't want any dynamic changes I just want to set the increment to 25, by the way English isn't my native language so please don't judge me

gfvalvo, 2022-12-19:
The library is written to be customizable by creating a class that inherits from the NewEncoder class. This is illustrated in the "CustomEncoder" example where an encoder class that implements "wrap around" behavior is demonstrated. Here's a simple example that allows a user-specified increment value:
```
#include "Arduino.h"
#include "NewEncoder.h"

// Demonstrate creation of custom Encoder using inheritance.
// This encoder "wraps around" when it hits the min or max limit

class CustomEncoder: public NewEncoder {
  public:
    CustomEncoder() :
      NewEncoder() {
    }
    CustomEncoder(uint8_t aPin, uint8_t bPin, int16_t minValue, int16_t maxValue, int16_t initalValue, uint16_t delta = 1, uint8_t type = FULL_PULSE) :
      NewEncoder(aPin, bPin, minValue, maxValue, initalValue, type), countDelta(delta) {
    }
    virtual ~CustomEncoder() {
    }

  protected:
    virtual void updateValue(uint8_t updatedState);

  private:
    int16_t countDelta;
};

void ESP_ISR CustomEncoder::updateValue(uint8_t updatedState) {
  if ((updatedState & DELTA_MASK) == INCREMENT_DELTA) {
    liveState.currentClick = UpClick;
    if (_maxValue - liveState.currentValue >= countDelta) {
      liveState.currentValue += countDelta;
    } else {
      liveState.currentValue = _maxValue;
    }
  } else if ((updatedState & DELTA_MASK) == DECREMENT_DELTA) {
    liveState.currentClick = DownClick;
    if (liveState.currentValue - _minValue >= countDelta) {
      liveState.currentValue -= countDelta;
    } else {
      liveState.currentValue = _minValue;
    }
  }
  stateChanged = true;
}

// Pins 2 and 3 should work for many processors, including Uno. See README for meaning of constructor arguments.
// Use FULL_PULSE for encoders that produce one complete quadrature pulse per detnet, such as: https://www.adafruit.com/product/377
// Use HALF_PULSE for endoders that produce one complete quadrature pulse for every two detents, such as: https://www.mouser.com/ProductDetail/alps/ec11e15244g1/?qs=YMSFtX0bdJDiV4LBO61anw==&countrycode=US&currencycode=USD
CustomEncoder encoder(2, 3, -50, 50, 0, 7, FULL_PULSE);
int16_t prevEncoderValue;

void setup() {
  CustomEncoder::EncoderState state;

  Serial.begin(115200);
  delay(2000);
  Serial.println("Starting");
  if (!encoder.begin()) {
    Serial.println("Encoder Failed to Start. Check pin assignments and available interrupts. Aborting.");
    while (1) {
      yield();
    }
  } else {
    encoder.getState(state);
    Serial.print("Encoder Successfully Started at value = ");
    prevEncoderValue = state.currentValue;
    Serial.println(prevEncoderValue);
  }
}

void loop() {
  int16_t currentValue;
  CustomEncoder::EncoderState currentEncoderState;

  if (encoder.getState(currentEncoderState)) {
    currentValue = currentEncoderState.currentValue;
    if (currentValue != prevEncoderValue) {
      Serial.print("Encoder: ");
      Serial.println(currentValue);
      prevEncoderValue = currentValue;
    }
  }
}
```

gfvalvo, 2023-01-27:
I have not heard back from [kakhi-kotchauri](https://github.com/kakhi-kotchauri). Issue closed.

closed 2023-01-27 by gfvalvo