gfvalvo/NewEncoder #23: How can I add this library to PlatformIO vscode?

Salty-Doug, 2023-01-12:
Hi,
I have started using PlatformIO in vscode a few weeks ago and I really prefer the interface over Arduino IDE. I am working on an ESP8266 project that reads two encoders and displays the values on two 7-Segment MAX7219 displays.  I successfully used this library in the Arduino IDE but I would like to migrate to PIO.  I do not see this library available in PIO.  Can I manually add it? 

Thanks in advance.

Salty Doug

gfvalvo, 2023-01-16:
Sorry, I don't know as I've never used PlatformIO. With [Eclipse / Sloeber](https://eclipse.baeyens.it/) I just provide a path to my custom library folder. How did you install it for the Arduino IDE?

Salty-Doug, 2023-01-17:
This library was listed in the Library Manager of Arduino IDE.  Arduino IDE will open and add a .zip file from GitHub, but I don’t see that functionality in PlatformIO.

gfvalvo, 2023-01-17:
> This library was listed in the Library Manager of Arduino IDE.

This library is not listed in the Arduino IDE's Library Manager. Perhaps you're thinking of the [PJRC Encoder Library](https://www.pjrc.com/teensy/td_libs_Encoder.html)? Or some other? Regardless, I'm not at all familiar with PlatformIO. Why don't you just download the .h and .cpp file and use it in your project that way?

Salty-Doug, 2023-01-18:
I found a way to add a library manually: "pio lib -g install (path)".

closed 2023-01-18 by Salty-Doug