gfvalvo/NewEncoder #27: Update to allow use of 6 external interrupt pins on the ATMega boards

gndprx, 2023-03-01:
Before I fork this and start hacking up the library, are there plans to add a version that supports all 6 external interrupts on the Mega board?  Interrupts 0-5 on pins 2, 3, 21, 20, 19, 18 respectively.

I'm using this library for an encoder wheel stepper motor driver and I'd love to be able to connect multiple steppers and wheels to a single board which in theory the Mega should support.

Thanks for all the hard work.

gndprx, 2023-03-01:
I guess in theory I could just instantiate 3 instances of NewEncoder and just pass in the different interrupt pins. That would be easier than modifying the library unless there is something in there that prevents the use of a Mega board or pins other than 2/3

gfvalvo, 2023-03-08:
> Before I fork this and start hacking up the library, are there plans to add a version that supports all 6 external interrupts on the Mega board?
As far as I know, it already does (but I don't have a Mega to test it). Have you had a problem running 3 encoders on a Mega? Did you try the "MultipleEncoders.ino" example and modify it for 3 encoders with the appropriate pin numbers?

gfvalvo, 2023-03-08:
> I could just instantiate 3 instances of NewEncoder

That's the intended technique. OOP!!!

gfvalvo, 2023-03-14:
Closing issue. @gndprx, repost if you continue to have problems with using 3 encoders on a Mega.

closed 2023-03-14 by gfvalvo

gndprx, 2023-03-14:
Yep sorry. Good to close.  Thanks.