gfvalvo/NewEncoder #31: Support for type quarter pulse

chitz852, 2024-04-08:
Hi, it looks like there is another type of encoder that needs 4 detents. could please add support for it ?
Example Grayhill SERIES 61C encoder https://www.farnell.com/datasheets/110579.pdf
[image]
Btw that's the encoder on the bmw e46 radio small monochrome navigation system 65126976899.

gfvalvo, 2024-04-08:
Are you absolutely sure that each of the “Positions” in that diagram corresponds to a detent stop? If so, I don’t see how to debounce this with the algorithm used by NewEncoder. Image you’re sitting at “Position 2” in that diagram, and you see a rising edge on Output B. Fine, that’s one click clockwise. But, then you see Output B fall again. That could be from contact bounce or an intentional click back counter-clockwise. I don’t know how to tell the difference.

I don’t have one of these devices to test any implementation.

chitz852, 2024-04-10:
Verified electrically and with the HALF_PULSE I get it working every second detent so yes. But you're right the library may be an overkill for this type. I can get its position at any given time by reading the pins, and knowing the last position I know in which direction it goes too. I assume because it's optical the "contacts" don't bounce therefore no need for the debounce logic.

gfvalvo, 2024-04-23:
Unfortunately, this type of decoder is not supported by the State Table algorithm used by NewEncoder.

closed 2024-04-23 by gfvalvo