gfvalvo/NewEncoder #33: IDE/Wokwi naming conflict with https://github.com/alexitoo00/NewEncoder

drf5n, 2025-01-25:
I tried to use the examples in Wokwi, which uses the same library-discovery scheme as the IDE, and ended up installing the https://github.com/alexitoo00/NewEncoder library, which has a different API.

gfvalvo, 2025-04-15:
I'm not familiar with Wokwi's library system. You should raise this issue with the developers of that platform.

closed 2025-04-15 by gfvalvo

drf5n, 2025-04-17:
I don't think it's a Wokwi issue, Arduino points to the alexitool00 library:

https://docs.arduino.cc/libraries/newencoder/  ->  https://github.com/alexitoo00/NewEncoder

and the IDE:

[image]

Your suggestion to install a sketch-local copy of a library that name-clashes with an Arduino registered library and use `#include "NewEncoder.h"` will indeed work.

gfvalvo, 2025-04-17:
> Your suggestion to install a sketch-local copy of a library that name-clashes with an Arduino registered library and use `#include "NewEncoder.h"` will indeed work.

You could also simply not install the alexitool00 library at all. Just install this one in the IDE via the .zip file.

drf5n, 2025-04-17:
I guess I'm fussing about the name-clash with an registered library.  If your library needs a non-standard installation to work around the IDE library system, then at the very least, the  `#include "NewEncoder.h"` lines in the examples should point to the repository:

```
#include "NewEncoder.h" //  https://github.com/gfvalvo/NewEncoder
```

...so that folks with a sketch or snippet of the code can get a hint to figure out how to avoid the errors.

gfvalvo, 2025-04-17:
> I guess I'm fussing about the name-clash with an registered library. If your library needs a non-standard installation to work around the IDE library system, then at the very least, the `#include [...]
The example itself is from the repository. So it stands to reason that you don't need a pointer to it. Regardless, if you fell strongly enough about this, feel free to fork repository, rename it, and submit for inclusion as a registered library.